werasik2aa/Huawei-Unlock-Tool #4: Is there any possibility to support SOC for Kirin 970 and above?

Coconutat, 2023-06-05:
I don't know how those dongle software works, but from the feature list, it seems that the release of the Bootloader lock on SoCs from 970 to 980 is forcibly released in TEST POINT 1.0 mode.They pass by the read  bootloader code and use it.  
So,maybe we can unlock BL lock though TESTPOINT 1.0.  
USER LOCK is another issuse.  
If Kirin SOC unlock BL LOCK and USER LOCK, it can bypass upgrade mode or Hi Suite mode and freely upgrade and descend the system. And there's no need to worry about BL LOCK being locked again.

werasik2aa, 2023-06-06:
https://github.com/mashed-potatoes/PotatoNV
ask this question to. because i just mergered their tool with this...

Coconutat, 2023-06-06:
> Ask this question to https://github.com/mashed-potatoes/PotatoNV. because i just mergered their tool with this... and they have this warning
> 
> 100% incompatible CPUs with PotatoNV: Kirin 710, 710A, 710F, 810, 970, 980, 985, 990 & newer.

OK,I get it.

werasik2aa, 2024-01-08:
Now added more device support