werasik2aa/Huawei-Unlock-Tool #7: Adding support to G7-L01

DerivativeOfLog7, 2023-06-29:
I currently own one of those and I'd be glad to help

werasik2aa, 2023-06-30:
You have fully unlocked phone?

DerivativeOfLog7, 2023-06-30:
Nope... But I do know it's possible to get root by downgrading to K and using exploits

werasik2aa, 2023-06-30:
You know how to root it hmmm. So i think huawei don't have big protection... But

DerivativeOfLog7, 2023-06-30:
Well 4.4.4 is really old by now, 5 is also technically compatible with dirtycow but I can't get a root shell with it

DerivativeOfLog7, 2023-06-30:
> You know how to root it hmmm. So i think huawei don't have big protection... But
> https://androidhost.ru/7Da

Yes! I tried taking a look at the source and the files that are downloaded for QC unlocks, and try to compare it with that board software, but couldn't get much out of it

werasik2aa, 2023-06-30:
If this firmware which give match to device, try to upload aboot.mbn, sbl1.mbn 
Or emmc appsboot.mbn to phone by edl9008

DerivativeOfLog7, 2023-07-01:
> If this firmware which given to you match with your device, try to upload aboot.mbn, sbl1.mbn
> Or emmc appsboot.mbn as aboot to phone by edl9008

How can I do that? I tried toying with [edl](https://github.com/bkerler/edl), on Windows I can't get it to work at all, I tried with the live ISO and it launches, but after uploading the loader I either get no output or errors like USB Overflow (I even tried using the G7 loader from this project but nada)

Right now I'm making a backup of the flash using this program

DerivativeOfLog7, 2023-07-01:
OK I figured out how to write the images... and the phone now boots straight to fastboot mode. What now?

werasik2aa, 2023-07-01:
sorry for long answer. Test the fastboot flash recovery some.img

DerivativeOfLog7, 2023-07-02:
Nope... not unlocked
fastboot flash gives command not allowed, and if I try to flash a custom recovery via TWRP it refuses to boot it.

I was wondering, since we can read/write apparently the whole flash via edl, how difficult would it be to do something like PotatoNV, overwriting the unlock code with our own?

werasik2aa, 2023-07-02:
there's no function in function list of aboot's fastboot... about rewrite code

DerivativeOfLog7, 2023-07-02:
> there's no function in function list of aboot's fastboot... about rewrite code

But since we have r/w access to the whole flash, couldn't we just overwrite it that way? Unless it's not calculated on the fly or something
I'm just throwing ideas, unfortunately I don't know anything about REing code

werasik2aa, 2023-07-06:
Try to use gpt reader. I mean partitions read... Try find nvme or oeminfo maybe inside should be code. 4.4 has old aboot i think

werasik2aa, 2023-11-24:
> But since we have r/w access to the whole flash, couldn't we just overwrite it that way? Unless it's not calculated on the fly or something
> I'm just throwing ideas, unfortunately I don't know anything about REing code

it's possible to rewrite key, but need to use huawei certs for that.

werasik2aa, 2023-11-30:
I don't wan't to bring here root methods to this tool.
Maybe it's will be soon. But not today. For many devices needs an unlocked bootloader.

werasik2aa, 2024-01-05:
Added

werasik2aa, 2024-01-05:
I think should work

werasik2aa, 2024-01-08:
https://github.com/werasik2aa/UnlockedHuaweiBootloader/issues/1

werasik2aa, 2024-01-08:
Closed here, not inside

closed 2024-01-08 by werasik2aa

hicode002, 2024-11-17:
> > 但是，既然我们有对整个 Flash 的 r/w 访问权限，我们就不能用这种方式覆盖它吗？除非它不是动态计算的，或者我只是在抛出想法，不幸的是，我对重新编写代码一无所知
> 
> 可以重写密钥，但需要使用 Huawei Certs。所有手机都相同

No.Huawei did not configure secure boot  correctly(pbl not fused),which means that we can patch sbl1 to disable the aboot signature check.

Remember huawei uses its own method to verify aboot in sbl1,and it does not use qualcomm certs.(So we can use qtestsign tool to generate a test key cert)