werasik2aa/Huawei-Unlock-Tool #13: System.Collections.Generic.KeyNotFoundException: Данный ключ отсутствует в словаре.

hkvert, 2023-12-21:
[image]
************** Текст исключения **************
System.Collections.Generic.KeyNotFoundException: Данный ключ отсутствует в словаре.
   в System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   в HuaweiUnlocker.Window.UNLBTN_Click(Object sender, EventArgs e)
   в System.Windows.Forms.Control.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   в HuaweiUnlocker.UI.NButton.OnMouseUp(MouseEventArgs e)
   в System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ButtonBase.WndProc(Message& m)
   в System.Windows.Forms.Button.WndProc(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Загруженные сборки **************
mscorlib
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4614.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
HuaweiUnlock
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///C:/RELEASE/HuaweiUnlock.exe
----------------------------------------
System.Windows.Forms
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4550.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4605.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Core
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4605.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Configuration
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Windows.Forms.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------
System.Management
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4435.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
mscorlib.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------

werasik2aa, 2023-12-21:
Хм сейчас, посмотрю возможно я привел не верную ссылку

werasik2aa, 2023-12-21:
Попробуй ещё раз, я обновил ссылки

werasik2aa, 2023-12-21:
Возможно у тебя сайта нет в списках, или произошла ошибка... Попробуй удалить файлы languages

werasik2aa, 2023-12-21:
Сейчас глянем, почему так..
А пока попробуй через qfill
https://github.com/werasik2aa/UnlockedHuaweiBootloader
AUM-L41.ZIP

werasik2aa, 2023-12-21:
Исправил ссылку, теперь должно работать!

werasik2aa, 2023-12-21:
[image]

hkvert, 2023-12-21:
Да, теперь работает, но выдает:

```
[INFO] Trying to unlock:  AUM-L41(FULL_METHOD2)
[INFO] Trying to unlock:  Reading GPT... Please wait
[INFO] DONE 21.12.2023 19:44:11
[ERROR] Error Reading GPT! Wrong firehose or EMMC GPT on device empty! 
[ERROR] FAILED TO UNLOCK BOOTLOADER 
```

на 4пда почитал:

```
Q9. Error Reading GPT! Wrong firehose or EMMC GPT on device empty!
A9. Попробуйте драйвера старее. Попробуйте пере-зайдите в EDL. Либо загрузчик выбирается не правильно, выберите его вручную.
Если не помогает возможно на вашем устройстве нет разметки gpt.
```

Качал кучу разных драйверов, пере-заходил в edl, но не помогло. Как понимаю у меня вариант "на вашем устройстве нет разметки gpt", и вопрос, как эту разметку сделать?

werasik2aa, 2023-12-21:
А устройство было с разметкой до этого?

hkvert, 2023-12-21:
> А устройство было с разметкой до этого?
без понятия

werasik2aa, 2023-12-21:
зайдите в раздел partitions, и нажмите read gpt

hkvert, 2023-12-21:
[INFO] Checking connection... 
[INFO] Reading GPT... Please wait 
[INFO] DONE 21.12.2023 20:11:13
[ERROR] Error Reading GPT! Wrong firehose or EMMC GPT on device empty!

werasik2aa, 2023-12-21:
я думаю, firehose в папке с программой либо отсутствует, либо я не тот положил
[image]
должно быть так

werasik2aa, 2023-12-21:
примерно, правда у меня нет aum

werasik2aa, 2023-12-21:
посмотрите в диспетчере задач, нет ли программ занимающие порт
emmcdl.exe
fh_loader.exe

werasik2aa, 2023-12-21:
Если устройство изначально не грузилось или нуждается в восстановлении используйте board
https://drive.google.com/file/d/1lMvL3xUohLRENhgO5j_FvdzMRmn9jerD/edit
а потом накатите сервисную прошивку

werasik2aa, 2023-12-21:
Если не помогает попробуйте этот загрузчик
[Snapdragon430.zip](https://github.com/werasik2aa/Huawei-Unlock-Tool/files/13744678/Snapdragon430.zip)

werasik2aa, 2023-12-21:
Я обновил программу в V20F, в архив добавил AUM41, должно заработать

hkvert, 2023-12-21:
> Я обновил программу V20F, в архив добавил AUM41 теперь, должно заработать, https://github.com/werasik2aa/Huawei-Unlock-Tool/releases/tag/20

[INFO] Checking connection... 
[INFO] Reading GPT... Please wait 
[INFO] DONE 21.12.2023 20:47:35
[ERROR] Error Reading GPT! Wrong firehose or EMMC GPT on device empty!

werasik2aa, 2023-12-21:
Рекомендую выставить в программе вот так:
[image]
[image]

hkvert, 2023-12-21:
в программе выставлено все так же как на скринах, и все равно ошибка [ERROR] Error Reading GPT! Wrong firehose or EMMC GPT on device empty!

werasik2aa, 2023-12-21:
Странно очень. попробуйте отключить аккумулятор, потом войти в edl

hkvert, 2023-12-21:
Аккумулятор всегда отключаю перед тем как заходить в edl

werasik2aa, 2023-12-21:
Попробуйте оставить графу загрузчика пустую
[image]
как здесь.

werasik2aa, 2023-12-21:
[image]
проверил на atu-l31

hkvert, 2023-12-21:
> Попробуйте оставить графу загрузчика пустую ![image](https://private-user-images.githubusercontent.com/20810116/292283013-a4761599-f286-4184-a637-8c479926a329.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxODIyMDcsIm5iZiI6MTcwMzE4MTkwNywicGF0aCI6Ii8yMDgxMDExNi8yOTIyODMwMTMtYTQ3NjE1OTktZjI4Ni00MTg0LWE2MzctOGM0Nzk5MjZhMzI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIxVDE4MDUwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1NDcxM2MzZWM2NGYxOGQ3ZGRmYmQzOWFhYjYyMTA5Mjc1Y2U3MWQwMmU4YmZhNzk3ZTI4MGQzNzdmYmRjNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Klhe9pBX7AygbtWecDBxkjyoQFdDznFxpjatAdeE5lk) как здесь. И нажмите разблокировку. Программа должна выбрать любой подходящий примерно так

[INFO] Checking connection... 
[INFO] DONE 21.12.2023 21:06:21
[INFO] HW_ID:  NaN
[INFO] SW_ID:  NaN
[INFO] OEM_PK_HASH:  NaN
[INFO] SBL_VER:  NaN
[INFO] CPU:  ERROR: FAILED UNKNOWN ERROR
[INFO] Device cannot be readed! Please reconnect it!

werasik2aa, 2023-12-21:
у тебя устройство видно, не читается, 
должно быть так
[INFO] HW_ID:  0x000560e1
[INFO] SW_ID:  NaN
[INFO] OEM_PK_HASH:  0x6bc369511da9cadb3a7af61574f89db385003d6241bdd1ff573dba61bf6ae119
[INFO] SBL_VER:  0x00000000
[INFO] CPU:  MSM8917 
примерно

werasik2aa, 2023-12-21:
[INFO] Checking connection... 
[INFO] DONE 22.12.2023 1:08:44
[INFO] HW_ID:  0x000560e1
[INFO] SW_ID:  NaN
[INFO] OEM_PK_HASH:  0x6bc369511da9cadb3a7af61574f89db385003d6241bdd1ff573dba61bf6ae119
[INFO] SBL_VER:  0x00000000
[INFO] CPU:  MSM8917
[INFO] Started unlock process 
[INFO] Connected PORT:  Qualcomm HS-USB QDLoader 9008 (COM25)
[INFO] Flashing Loader:  qc_boot\AGS\Qualcomm Snapdragon 425.mbn
[INFO] DONE 22.12.2023 1:08:47
[INFO] Trying to unlock:  ATU-L31-L21-LOWER(FULL)
[INFO] Please wait:  99%
[INFO] DONE 22.12.2023 1:08:48
[INFO] Trying to Write:  aboot
[INFO] Please wait:  75%
[INFO] DONE 22.12.2023 1:08:50
[INFO] Trying to Write:  kernel
[INFO] DONE 22.12.2023 1:09:48
...
Лог успешной разблокировки должен быть примерно таков

werasik2aa, 2023-12-21:
для проверки подключения устройства нажмите.
Оно должно предложить подходящие загрузчики
[image]

hkvert, 2023-12-21:
отключил подпись драйверов, пере-воткнул в другой порт:
```
[INFO] HW_ID:  NaN
[INFO] SW_ID:  NaN
[INFO] OEM_PK_HASH:  NaN
[INFO] SBL_VER:  NaN
[INFO] CPU:  ERROR: FAILED UNKNOWN ERROR
[INFO] Trying to guess firehose for this device... 
[INFO] Device cannot be readed! Please reconnect it! 
```
 visual c++ redist стоят 
[image]

werasik2aa, 2023-12-21:
в диспетчере устройств восклицательный знак горит?

hkvert, 2023-12-21:
[image]

werasik2aa, 2023-12-21:
[image]
странно очень почему не читается

werasik2aa, 2023-12-21:
зайди в диспетчер задач, взгляни нет ли там emmcdl или fh_Loader.exe

hkvert, 2023-12-21:
нету

werasik2aa, 2023-12-21:
вот это уже странно. Обычно emmcdl читает любой 
[image]
Попробуй через qfill прошить...
Там есть менеджер разделов

hkvert, 2023-12-21:
```
Version 2.16
PblHack: Error - 4
Did not receive Sahara hello packet from device
Status: 6 The handle is invalid.
```

werasik2aa, 2023-12-21:
https://4pda.to/forum/index.php?showtopic=753902&view=findpost&p=127020635
тут есть драйвер,

hkvert, 2023-12-21:
Винда 10 с сайта майкрософт, сейчас попробую

werasik2aa, 2023-12-21:
Если с сайта майки, то скорее всего драйвер.

werasik2aa, 2023-12-21:
[image]
а ещё может быть такое на ноутбуках

werasik2aa, 2023-12-21:
[image]
А ещё у меня время близится к глубокой ночи, получится завтра продолжить работу над проблемой?
Или срочное дело?

hkvert, 2023-12-21:
поставил тот драйвер, перезагрузил пк.  тел перевел в edl 
```
C:\RELEASE\Tools>emmcdl -p COM4 -info
Version 2.16
PblHack: Error - 4
Did not receive Sahara hello packet from device
Status: 6 The handle is invalid.
```

hkvert, 2023-12-21:
> ![image](https://private-user-images.githubusercontent.com/20810116/292288601-a2dbbe9c-7dd1-4dc7-8b8d-b32ae1ed8433.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxODM5MDksIm5iZiI6MTcwMzE4MzYwOSwicGF0aCI6Ii8yMDgxMDExNi8yOTIyODg2MDEtYTJkYmJlOWMtN2RkMS00ZGM3LThiOGQtYjMyYWUxZWQ4NDMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIxVDE4MzMyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0N2Y0OGRiZGQ1ZTE5ZjgyYWQ3Yjk4ZjMwZTdhODY2ZTk5Zjg3MTBjMjA1NDNjZjMwODllMTdlNjZhZDViNjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VlGT0HFjyqLXYV5c11GQhVky75DORcxKJQUOtayHV-Q) А ещё у меня время близится к глубокой ночи, получится завтра продолжить работу над проблемой? Или это срочное дело?

а, давайте тогда завтра продолжим

werasik2aa, 2023-12-21:
Ну я ещё могу 30 мин уделить

werasik2aa, 2023-12-21:
>  пк. тел перевел в edl

попробуй это

> ![image](https://private-user-images.githubusercontent.com/20810116/292287396-8414a1c8-4f17-48c0-8997-78dddd2177e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxODM2NzIsIm5iZiI6MTcwMzE4MzM3MiwicGF0aCI6Ii8yMDgxMDExNi8yOTIyODczOTYtODQxNGExYzgtNGYxNy00OGMwLTg5OTctNzhkZGRkMjE3N2U2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIxVDE4MjkzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3NjNhNDMzODE4MjliYzUyOTE5YzVlYjJkODEzYWIxZmQzNTkyMGI5MzdmMzRmZWVlYTgzNjg5YWVlNWQ5Y2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CUZdc-qMMv4CjPTPh40nGKd1BKGm0VaRN1QFjF0gQ38) а ещё может быть такое на ноутбуках, Нужно отключать

hkvert, 2023-12-21:
>  а ещё может быть такое на ноутбуках, Нужно отключать

у меня компьютер, но это на всякий выключил, и все равно выдало ту ошибку

werasik2aa, 2023-12-21:
Тогда, куда более странно. Устройство может заходить в рекавери?

werasik2aa, 2023-12-21:
Попробуй запустить от имени администратора, некоторым помогает...

hkvert, 2023-12-21:
> Тогда, куда более странно. Устройство может заходить в рекавери? Или только входит в dload 9008?

и в рекавери и в фастбут спокойно заходит

hkvert, 2023-12-21:
иногда кстати устройство заходит в странный режим и показывает 3 порта 
[image]

werasik2aa, 2023-12-21:
Вот это куда странно. Программа не может соеденится с устройством.
А ошибка вообще на стадии pbl hack...

Попробуйте использовать не usb3.0 а usb2.0. или наоборот. Т.к эта ошибка как я знаю выходит по трем причинам
1. Драйвера
2. Порт занят
3. Не правильное соединение testpoint.а

werasik2aa, 2023-12-21:
https://disk.yandex.ru/d/rQ4e72f1FhbuWg/TESTPOINTS
Посмотрите тестпоинт

werasik2aa, 2023-12-21:
Ладно, у меня время закончилось, а то спички в глаза буду вставлять. Завтра попытаем счастье заставить ваше устройство прошиваться.

А ещё попробуйте сменить кабель...

werasik2aa, 2023-12-21:
> иногда кстати устройство заходит в странный режим и показывает 3 порта [...]

Этот режим называется manufacture. Для чтения кода донгд коробкой.

werasik2aa, 2023-12-21:
Если необходимо можете попросить https://4pda.to/forum/index.php?showuser=6811229 помочь.
Или
https://4pda.to/forum/index.php?showuser=2988300 (в профиле телеграм)

werasik2aa, 2023-12-22:
Вновь приветствую тебя...

hkvert, 2023-12-22:
> Вновь приветствую тебя...

👋

werasik2aa, 2023-12-22:
Получилось что-то?

hkvert, 2023-12-22:
> Получилось что-то? Пока меня не было? Или не пытался :D
Не не пытался. Домой только час назад пришел.

werasik2aa, 2023-12-22:
хм, будем пробовать?

[29 more replies]

closed 2023-12-25 by hkvert