werasik2aa/Huawei-Unlock-Tool #16: Need ANE-LX1 Compatibility

lyonelf, 2023-12-29:
Hi, 

is it possible to have this device listed?

Thanks a lot in advance

werasik2aa, 2023-12-29:
if it's HiSilicon Kirin 659
you can try potatonv or try to write the loaders from https://github.com/werasik2aa/UnlockedHuaweiBootloader/tree/main/UnsortedUntestedKirins
throught "Huawei Flash Tools IDT"
Kirin 659

lyonelf, 2023-12-29:
yes it's 659

Potato need to open the back and pin shortcircuit. It's not a 5mns méthod 

Your Unsorted link mark a 404 error

werasik2aa, 2023-12-29:
today i sorted it, https://github.com/werasik2aa/UnlockedHuaweiBootloader

werasik2aa, 2023-12-30:
So only one way to try to unlock device -> Use it or potatonv with open back of device.

werasik2aa, 2024-01-07:
You can try to downgrade to emui 9/8 and try to unlock with your own risk

lyonelf, 2024-01-07:
Yes but for a c109 version, i don't know if i can put a c432 Europe ROM.

And i didn't found a perfect tutorial to be confident during the manipulations

werasik2aa, 2024-01-08:
Hmm so you still have that device and can you access to testpoint?

werasik2aa, 2024-01-08:
And this phone has and emui 9/8?

werasik2aa, 2024-01-15:
closed due no activity

closed 2024-01-15 by werasik2aa