werasik2aa/Huawei-Unlock-Tool #17: PCUI & DBADAPTER

A2B1C4, 2024-01-02:
When I click read info, I am getting this error:
[WARN] Not Found PCUI 
[WARN] Not Found DBADAPTER

werasik2aa, 2024-01-02:
Because there's no read bootloader code functions. This function just for test. U can unlock your bootloader via 9008 mode

werasik2aa, 2024-01-02:
Using button Download And Unlock...

werasik2aa, 2024-01-02:
how to use (Old video)
https://www.youtube.com/watch?v=Hhmm2IOrbv4

werasik2aa, 2024-01-02:
yes

werasik2aa, 2024-01-02:
The first try to wipe data. If still bootlooop
 you need install .zip which decrypt the storage data,
 than you need wipe data in twrp

A2B1C4, 2024-01-02:
[no-verity-opt-encrypt-6.1.zip](https://github.com/werasik2aa/Huawei-Unlock-Tool/files/13812860/no-verity-opt-encrypt-6.1.zip)
Do you mean flashing this .zip using fastboot?

werasik2aa, 2024-01-02:
ys

werasik2aa, 2024-01-02:
twrp

A2B1C4, 2024-01-02:
I can't flash the recovery, looks like I made a mistake
[image]

A2B1C4, 2024-01-02:
[image]
See this one too, it might help.

werasik2aa, 2024-01-02:
for flash twrp u need to fastboot flash recovery_ramdisk #.img

A2B1C4, 2024-01-02:
OK, I will check the guide and I will tell you what happens after.

A2B1C4, 2024-01-02:
Good news, I flashed the recovery, bad news, the screen is not working.

werasik2aa, 2024-01-03:
Sorry for long answer because wes deep night. 
hmm, on my device atu-l31 it's working good.  maybe try:
1  Try to flash board firmware https://androidhost.ru/SEj
2 then flash the stock firmware via sd card/otg update
3 then try unlock device with tool. It's should help with unlocking

I know it's russian forum, but you can contact with https://4pda.to/forum/index.php?showuser=6811229 in QMS(Dm messages)
he has dongle box and can unlock your device with unlock code for free.

A2B1C4, 2024-01-03:
No problem, my device is ATU-L31, but it is international version, I flashed the recovery, the problem is that when TWRP recovery boots, the screen isn't working, I can't touch anything, I will try another TWRP recovery and tell you how it goes.

werasik2aa, 2024-01-03:
hm . Sometimes i have this error too.
You can use adb shell twrp
commands

A2B1C4, 2024-01-03:
Is there any non touch recovery for the ATU-L31? Because the twrp recovery is not responding to touch at all

werasik2aa, 2024-01-04:
try to use 
adb shell twrp sideload
adb sideload #.zip
adb shell twrp wipe data

werasik2aa, 2024-01-04:
you can controll twrp functions with adb shell
then use
twrp

A2B1C4, 2024-01-04:
I booted to edl, but the link is dead, I cant download it from that link, I also don't have a way to install the board firmware.

A2B1C4, 2024-01-04:
[image]
I am also seeing this when I flash the recovery.

werasik2aa, 2024-01-04:
for download the image need to login in .

werasik2aa, 2024-01-04:
for install booard software you need
1. qfill
2. Tool for short pins on
3. And firmware

For download the images from 4pda need to log in

A2B1C4, 2024-01-04:
I flashed one GSI rom using fastboot, and the twrp recovery magically worked, but the gsi rom is stuck on boot screen and cant conitune to system, I want to try android GSIs roms, and so far, none of them work, all of them are stuck on the boot screen.

werasik2aa, 2024-01-04:
You need to write vendor andr9 for gsi 9

werasik2aa, 2024-01-04:
https://drive.google.com/drive/mobile/folders/1aFTTMSArwBuF5tsJBkWJ7f_d7IQc7zyy
Vendor

And with current unlock don't try to flash a custom kernel! The kernel.img patched for avoid bootloop on gsi's rom.

A2B1C4, 2024-01-04:
Should I install the kernel and vendor after flashing the gsi?

werasik2aa, 2024-01-04:
You must install onky vendor. Don't touch kernel

A2B1C4, 2024-01-04:
Should I flash the vendor after flashing the gsi image or before flashing the gsi image?

werasik2aa, 2024-01-04:
You xan flash vendor after as well as you can flash it before. Just need to so e fix size of vendor in twrp if its won't be flashed

A2B1C4, 2024-01-04:
The touch screen is not working in twrp again, so I will try to flash using fastboot and I will comment what happens next.

A2B1C4, 2024-01-04:
I used fastboot flash vendor VENDOR.img, output is failed(remote: 'size too large')

werasik2aa, 2024-01-04:
Hmm for workin touch in twrp try
Fastboot erase userdata
Fastboot format userdata
Fastboot -w

A2B1C4, 2024-01-04:
I will try and I will comment any updates here.

werasik2aa, 2024-01-05:
Ok:)

A2B1C4, 2024-01-05:
I am getting this error when flashing vendor:
fastboot flash vendor VENDOR.img
Sending 'vendor' (507596 KB)                       OKAY [ 16.864s]
Writing 'vendor'                                   FAILED (remote: 'size too large')
fastboot: error: Command failed

werasik2aa, 2024-01-05:
hm strange

werasik2aa, 2024-01-05:
you need use an Hut and flash vendor throught gpt manager

A2B1C4, 2024-01-05:
The biggest problem for me is the touch screen problem, I don't want to go back to the stock firmware, if the touch works on twrp, then I can flash the vendor image using the twrp, fixing the touch problem is the most important thing for me.

werasik2aa, 2024-01-05:
For fix touch sometimes helps
fastboot erase userdata
fastboot erase cache
but...

A2B1C4, 2024-01-05:
Also, is there a way to install system and vendor images using the adb shell commands?

werasik2aa, 2024-01-05:
hmm only .zip 
adb shell twrp sideload
adb sideload somesystemrom.zip

A2B1C4, 2024-01-05:
Is there any other custom recovery for the ATU-L31 that is not-touch recovery?

werasik2aa, 2024-01-05:
Nah exist 2 types on 4pda

werasik2aa, 2024-01-05:
[twrp_3.3.1_atomu_20190824.zip](https://github.com/werasik2aa/Huawei-Unlock-Tool/files/13844819/twrp_3.3.1_atomu_20190824.zip)
Try this

A2B1C4, 2024-01-05:
OK, I will sideload it

A2B1C4, 2024-01-05:
Nope, not working too, the phone keeps booting back into fastboot mode, and touch not working

werasik2aa, 2024-01-05:
Try to flash stock recovery and full ota, then try unlock again

A2B1C4, 2024-01-05:
How?

A2B1C4, 2024-01-05:
> Twrp is very weird, I kept holding the volume + volume - and power button, and I kept holding them until it restarted for like 5 times, and decided to boot to twrp, and it worked again, anyway, I would like to say that there is no vendor partition on the phone, at least it does not show up in the advanced wipe section

werasik2aa, 2024-01-05:
Hm strange thing, flash the stock recovery from fastboot and write full ota.
Then try again unlock

A2B1C4, 2024-01-05:
[image]

[image]

Look at these images, they might help

A2B1C4, 2024-01-05:
Where can I find recover_ramdisk and erecovery_ramdisk? Are they in the update.app?

werasik2aa, 2024-01-06:
Yes recovery_ramdisk and erecovery_ramdisk exist in Update.app

A2B1C4, 2024-01-06:
I flashed both files, but I can't boot to recovery now

werasik2aa, 2024-01-06:
why you can't boot to recovery?

werasik2aa, 2024-01-06:
hm maybe the firmware from another atu?

A2B1C4, 2024-01-06:
Where can I download the firmware?

werasik2aa, 2024-01-06:
https://androidhost.ru/search.html

werasik2aa, 2024-01-06:
[image]
write like that in search

[80 more replies]

closed 2024-01-08 by werasik2aa