werasik2aa/Huawei-Unlock-Tool #18: Kirin Unlocker doesn't work

KaroloBC, 2024-01-02:
Hello, i tried to unlock bootloader on Mate 20 Lite (EMUI 10, 10.0.0.270); tool crashes (doing nothing after uploading xloader) when uploading bootloader to phone.
Log:

=============UNLOCKER BL/FRP (KIRIN TESTPOINT)============= 
[INFO] Checking connection... 
[INFO] Parsing: UnlockFiles\KIRIN710\manifest.xml 
[INFO] TITLE CPU: Kirin 710 
[INFO] DIR: KIRIN710 
[INFO] Verifying images... 
[INFO] Uploading bootloader: KIRIN710 
[INFO] -xloader

KaroloBC, 2024-01-02:
yeah, my name on telegram is szczurek (without profile photo)

werasik2aa, 2024-01-02:
Otherwise you can test this what where i took that loaders
[Uploading PotatoNV-Extra.zip…]()

werasik2aa, 2024-01-02:
[Uploading PotatoNV-Extra.zip…]()
[image]
is that you?

KaroloBC, 2024-01-02:
no. maybe u can send me your name? it will be faster, i guess

werasik2aa, 2024-01-02:
[image]
try find

werasik2aa, 2024-01-02:
otherwise you can test it https://raw.githubusercontent.com/werasik2aa/Huawei-Unlock-Tool/HUT/POTATONVBETA_NOTMINE.zip

KaroloBC, 2024-01-02:
I sent DM to you on telegram.

werasik2aa, 2024-01-03:
i think we need 100% downgrade the firmware

KaroloBC, 2024-01-03:
yeah, but where we can get rollback package? on androidfilehost it doesnt
exist (i mean emui 9.1, because trying to downgrade directly to emui 9 ends
with software install failed, emui 9.1 downgrade only exists for singlesim
SNE-L01 or my model SNE-L21 but other regions). I will answer you on
Telegram after school.

werasik2aa, 2024-01-04:
https://androidhost.ru/search.html

werasik2aa, 2024-01-07:
Unlocked. 
[image]
SNE-LX1 9.0.1.164

closed 2024-01-07 by werasik2aa