werasik2aa/Huawei-Unlock-Tool #19: account was hijacked

Bakunetsuuuuu, 2024-01-08:
When I tried to Bootloader Unlock my Huawei tablet using this software, a Vietnamese person suddenly accessed my Google account. 
I was saved because I had a passkey prepared, but it seems like the software was accessed the moment I pressed the button, which makes me suspicious.
I can't say that this software is bad in general, but I am concerned about the moment I use it.
Just to be sure, I would like the developer to check whether there are any vulnerabilities in this software.

werasik2aa, 2024-01-08:
This software c# project which open source. It didn't hijack accounts.
The only one place where you can download it's github. Ready to compile from source.

Its flashing only .img files to device. Nothing more.

Devinfo.img and cleaned frp.img with edl9008 mode

Bakunetsuuuuu, 2024-01-08:
Considering this possibility, I initialized all the devices I used just to be safe.
The only mystery is that it happens as soon as you use this software.

Bakunetsuuuuu, 2024-01-08:
No, I think it's unlikely because I check to see if the URL is official before downloading.
I don't think this software is bad. It's possible that I was accidentally attacked by a zero-day attack...

werasik2aa, 2024-01-08:
Hm maybe. Because only github contains this tool. I'm never publish it on another direct links or sites.
I too thinking it's zero-day attack. So i'm closing this issue as it completed

closed 2024-01-08 by werasik2aa