werasik2aa/Huawei-Unlock-Tool #22: Support for Huawei M5 Mediapad CMR-W09 Kirin 960, any thoughts on if it's savable.

NebularNerd, 2024-02-05:
I'm trying to resurrect my potentially dead CMR-W09 and wanted to give your software a try, but I'm struggling to see it in the support device list. Am I missing something or is it currently not supported? 

I've tried PotatoNV and some other tools, I also have an 'engineer cable' and the cover off. I replaced the battery and the charger board as one or the other seems to have died a while ago. The best I can do is get a COM port to appear, if I try flashing under PotatoNv it looks to copy data across but then timesout on the reboot stage.

werasik2aa, 2024-02-07:
this device isn't in supported devices list

werasik2aa, 2024-02-07:
> PotatoNv it looks to copy data across but then timesout on the reboot stage.
PotatoNV has lower setted timeout time, so that give's error

NebularNerd, 2024-02-07:
> this device isn't in supported devices list

Thanks for confirming, I thought this might be the case. I see you have changed the title and lok forward to it possibly being included in a later release.

werasik2aa, 2024-02-10:
Try to use androidutility tool, it's free after use it try to downgrade os untill you give the whitescreen bootloader

NebularNerd, 2024-02-11:
This one [https://www.androidutility.com/](https://www.androidutility.com/)

werasik2aa, 2024-02-16:
yes

NebularNerd, 2024-02-21:
Downloaded it, and it kept just asking for a `Smart card reader` or press to get into a limited version. Nothing much happened or seemed to work for me, no GUI or anything.

werasik2aa, 2024-02-23:
smart card not needed for kirin operations

NebularNerd, 2024-02-24:
I'll try again, the app would not even open beyond the dialog box asking for the smart card reader.

NebularNerd, 2024-03-28:
[https://www.androidutility.com/]( https://www.androidutility.com/) is dead ☹️, and wayback has the site but no content. Do you have a copy you could upload here?

werasik2aa, 2024-04-06:
> https://www.androidutility.com/ is dead ☹️, and wayback has the site but no content. Do you have a copy you could upload here?

https://androidfilehost.com/?fid=11701882489785043575

NebularNerd, 2024-04-07:
Thanks for that, this download keeps telling me there is a new version and do I want to upgrade. Clicking no means the software does not run. Clicking yes takes me to a download which you can't unpack without a password.

[image]

werasik2aa, 2024-05-14:
Needs to find a newer means

NebularNerd, 2024-05-14:
> Needs to find a newer means

I found a working copy, but I think my tablet may be very dead. Any effort to get flashboot working just gives the same issues with 'Device descriptor' errors.

werasik2aa, 2024-05-19:
hm, i think cable need or power less. I mean dead battery.. uh i thing you understand

NebularNerd, 2024-07-01:
Does anyone reading this have a good collection of the CMR-W09 board softwares? I've been experimenting with some older ones and seem to get some reaction from the tablet, but might need a newer one to really make it behave.

felpcereti, 2024-07-30:
@NebularNerd 

To find firmware on firmware finder https://professorjtj.github.io/ you need to run (don't need to set up anything) hisuite proxy https://github.com/ProfessorJTJ/HISuite-Proxy
this site is shit it and obfuscates it's code for some reason but it works. You can deobfuscate it using https://obf-io.deobfuscate.io/

NebularNerd, 2024-07-30:
@felpcereti 

Thanks for the pointers. 🙂 I am able to get firmware's off that site fairly easily, if you open the file list link it gives you an xml, you can then copy the file name in the `spath` or `dpath` keys and replace the `filelist.xml` in the address bar and you'll get a valid download url. 

However, from downloading a **lot** of these do not seem to contain the board software which is a slightly different thing. I think from looking at the URL's and HiSuite Proxy I need a 'recovery' version of the firmware if such a thing is possible to get a url for.