werasik2aa/Huawei-Unlock-Tool #25: [Request] black screen after tool finishes to unlock bootloader

nanerb25, 2024-05-02:
my issue was that after the tool flashes the aboot and sbl1 when i reboot my device, it vibrates but no display but i can boot to fastboot mode and flash partitions, hopefully you can come up with a solution to this

Device Info:

Android: 8.1.0
EMUI Version: 8.2.0.036(07JH)
Phone Model: DUB-LX3(C636)
Android Security Patch Level: Febuary 1, 2020

werasik2aa, 2024-05-13:
Hi there. Long weeks ago i was busy

werasik2aa, 2024-05-13:
So you see vibration and black screen hmm

werasik2aa, 2024-05-13:
When you trying to turn on it shows any logo? Or

nanerb25, 2024-05-13:
No, the phone vibrates but no screen, and it looks like it boots directly to some sort of qcom diagnostics based on windows device manager,  but sometimes just infinity reboot, but i can boot it to fastboot mode by holding volume down but ofc no screen(no backlight, nothing on the screen at all), also no logo

werasik2aa, 2024-05-13:
hmm let's think why this happenes

werasik2aa, 2024-05-13:
i have another bl pack. we can try them

werasik2aa, 2024-05-13:
[DUB-TEST.zip](https://github.com/werasik2aa/Huawei-Unlock-Tool/files/15293401/DUB-TEST.zip)

nanerb25, 2024-05-19:
hello, i will try to flash the files u gave right now

nanerb25, 2024-05-19:
sadly after flashing via fastboot via flashing the incompatible image, the screen is still black

nanerb25, 2024-05-19:
it looks like aboot is incompatible

werasik2aa, 2024-05-19:
looks like not only aboot here cause this

werasik2aa, 2024-05-19:
there exist some firmware packages for qpst
https://huaweistockrom.com/huawei-y7-2019-prime-dub-lx3

we can try some. Need only to flash sbl1.mbn and aboot.mbn

nanerb25, 2024-05-19:
should i send here the firmware i have working here?

nanerb25, 2024-05-19:
i will send it anyways

https://mega.nz/file/w9BzRRYJ#Yn3iuqU0rbMpLDe_R5RZv09U9Itdh1K9IJeC60MzVnA

nanerb25, 2024-05-19:
aboot is the problem because if i just restore the aboot from the working firmware, screen actually lights up and boots

werasik2aa, 2024-05-20:
This firmware which you provided. Is it board firmware? Because im didn't download it yet

nanerb25, 2024-05-20:
its a update firmware but the update.app is already extracted

werasik2aa, 2024-05-23:
hm. Strange i can't find another board software. I think need to use one

werasik2aa, 2024-05-23:
way. to flash full bootloader chain

werasik2aa, 2024-05-23:
just with partition manager
[LX3.zip](https://github.com/werasik2aa/Huawei-Unlock-Tool/files/15412213/LX3.zip)

werasik2aa, 2024-05-23:
[image]
interesting things

werasik2aa, 2024-05-23:
maybe because board firmware has different gpt

werasik2aa, 2024-05-31:
so i think 100 % because different gpt. 
The device tries to boot it. But it's goes to black screen because different gpt regions

werasik2aa, 2024-07-02:
No activity. Not interested or i don't know why you stopped this fight with Huawei bootloader

closed 2024-07-02 by werasik2aa

nanerb25, 2024-07-08:
im sorry, its because i killed.... the device's battery and its screen by accident, and so i forgot this also and got myself a new phone :)