werasik2aa/Huawei-Unlock-Tool #27: SLA-L22 and AGS-W09

felpcereti, 2024-07-30:
To find firmware on firmware finder https://professorjtj.github.io/ you need to run (don't need to set up anything) hisuite proxy https://github.com/ProfessorJTJ/HISuite-Proxy
this site is shit it and obfuscates it's code for some reason but it works. You can deobfuscate it using https://obf-io.deobfuscate.io/
1. Device name + code name | `Huawei P9 lite mini (SLA-L22 C432 B176)` https://www.gsmarena.com/huawei_p9_lite_mini-8833.php
2. Device board firmware(optional) | I have got it from firmware finder and i'm not sure if it is it 
filelist.xml: http://update.dbankcdn.com/download/data/pub_13/HWHOTA_hotaMigrate_900_9/e6/v3/069a44f07b214cdca77bad7e5b512e9a/full/filelist.xml
update.zip: http://update.dbankcdn.com/download/data/pub_13/HWHOTA_hotaMigrate_900_9/e6/v3/069a44f07b214cdca77bad7e5b512e9a/full/update.zip
I have a "memory dump" from the program for this device
3. CPU | `MSM8917` Snapdragon 425

1. Device name + code name | `Huawei MediaPad T3 10 (AGS-W09 C100 B278)` https://www.gsmarena.com/huawei_mediapad_t3_10-8636.php#ags-w09
2. Device board firmware(optional) | I have got it from firmware finder and i'm not sure if it is it
this one is for AGS-W09 C100 **B271** and it is the closest i could find
filelist.xml: https://update.dbankcdn.com/download/data/pub_13/HWHOTA_hotaMigrate_900_9/af/v3/fe244cccf9a24a0da596be6e67beb995/full/filelist.xml
update.zip: https://update.dbankcdn.com/download/data/pub_13/HWHOTA_hotaMigrate_900_9/af/v3/fe244cccf9a24a0da596be6e67beb995/full/update.zip
I have 2 "memory dumps"
one is patched by HUT program and doesn't boot (clicked download bootloader and unlock)
the second one is fixed by me using the update.zip I have provided above (extracted, and patched all the files that i have found the program modified) but after that in "about phone" it says custom version CUSTC100D001
3. CPU | `MSM8917` Snapdragon 425

I would also greatly appreciate help with fresh flashing these devices (factory reset with keeping the imei)

werasik2aa, 2024-07-31:
imeis in fsg, fsc, oeiminfo, modem

felpcereti, 2024-07-31:
Thank you for help with factory flashing. Could you please elaborate how do you make these custom images? ghidra, ida pro?

werasik2aa, 2024-08-15:
> Thank you for help with factory flashing. Could you please elaborate how do you make these custom images? ghidra, ida pro?

None of that

werasik2aa, 2026-01-05:
AGS-LXX bootloader perfectly works with AGS-WXX

closed 2026-01-05 by werasik2aa