werasik2aa/Huawei-Unlock-Tool #33: Fastboot ERR

Lgndwilloske, 2024-10-24:
[WARN] Device cannot be readed! Please reconnect it!  [HISI] Maybe fastboot drivers not installed

werasik2aa, 2024-10-31:
First explain which device.
Then explain what you tried to do

Lgndwilloske, 2024-11-02:
hey, sorry for the long wait on the reply. I have a old Huawei Mate 20 Lite (SNE-LX1, KIRIN710) it has recently been locked due to frp and I latest the Google account. the Fastboot state is FBLOCKED, bootloader is locked and of course; it is FRP locked. when I tried to unlock the bootloader using the TP method, it returned with this error code. I am realy stuck and would appreciate any help I can get. I think the phone is running EMUI 12 and if what I researched I need to downgrade the firmware of which I do not know how to do, and I have researched but still not found anything.

werasik2aa, 2024-12-26:
Try downgrade. Use android utility tool for temp unlock

The-Chilliest-Willy-Ever, 2025-02-25:
do you know any tutorials for this? im really stuck

Micoder-dev, 2025-03-31:
Same issue