werasik2aa/Huawei-Unlock-Tool #35: Some ways to unlock old huawei devices with qualcomm soc and need help

hicode002, 2024-11-17:
For some old huawei devices,they did not configure secure boot correctly(not fused)，for example ,all devices with msm8916 msm8909 and some devices with msm8953.They just used the modified sbl to verify the signature of the Aboot partition header (using RSA), and then configured the bootloader lock inside the Aboot.
So we can directly patch sbl, skip the verification of aboot, and then use lk1st or Qualcomm's original lk to port the display panel driver, which can unlock the bootloader.

Here is an example.
https://wiki.postmarketos.org/wiki/Huawei_Ascend_G7_(huawei-g7)#Preparation

I have tried it ,but meet some problems.
https://github.com/msm8916-mainline/lk2nd/issues/446
could you help me??
thanks

werasik2aa, 2025-01-19:
NEED PBL  TO SKIP VERIFY OF SBL

werasik2aa, 2025-01-19:
https://alephsecurity.com/2018/01/22/qualcomm-edl-1/

werasik2aa, 2025-05-28:
Only new .mbns probally maybe can be used

closed 2025-05-28 by werasik2aa

hicode002, 2025-10-06:
> NEED PBL TO SKIP VERIFY OF SBL

For old devices,they are unfused.Just patch sbl and aboot!